endorphin3d/endorphin #1: Difficulty configuring marlin

powermandan, 2023-09-26:
Hi, I have implemented stage one of endorphin. I having problems with the firmware. I'm am no stranger to building marlin firmware.

When you build firmware like this:
#define MARKFORGED_XY
#define INVERT_X_DIR false
#define INVERT_Y_DIR false

X moves in correct direction when operated on its own. When you move Y it moves in correct direction. But the X moves as well. The X stepper is trying to compensate. But it turning the wrong way.

When you set it up like this 
#define MARKFORGED_XY
#define INVERT_X_DIR True
#define INVERT_Y_DIR false

The X moves in the opposite direction to expected. When you move Y it goes in correct direction and the X is adequately compensated and remains still on gantry. What this is means is the printer is now effectively mirrored. When I try to home it does to wrong end of gantry.

Any Ideas? As a last resort I am considering reprinting and moving X stepper to other side of printer. My suspicion is that it will work then. Although the maths make my head spin

endorphin3d, 2023-09-26:
Yes, it seems that a few users have experienced this same thing. It seems to be board-dependent as I've never had this issue. One Reddit user (u/Xyzjin) had this to say:
 
> So the correct settings (end stops and home position in the right back corner) for the Marlin Firmware are:
> 
> define INVERT_X_DIR false
> 
> define INVERT_Y_DIR true
> 
> define INVERT_Z_DIR false
> 
> define X_HOME_DIR 1
> 
> define Y_HOME_DIR -1
> 
> define Z_HOME_DIR -1
> 
> But this also comes with one thing to mention. The x/y grid (or starting point for all movements) is in this case rotated to the left back corner instead the left front corner like in the markforged schematic. Also the printed objects are mirrored. For example letters on the top surface are now right to left and flipped 180degrees.

So I would give those home direction settings a try and report back. If you find that works for you as well, I will update the docs to include this.

powermandan, 2023-09-27:
I reprinted the parts today and mirrored them to the other side.

In Marlin Inverted X

And I'm happy to report that home has moved back to the normal position and parts are not printed in mirror.

Thanks for taking the time to create this mod

SirNukalot, 2023-09-30:
> Yes, it seems that a few users have experienced this same thing. It seems to be board-dependent as I've never had this issue. One Reddit user (u/Xyzjin) had this to say:
>
>
> So I would give those home [...]

I've also think the reason behind you have to invert the X or Y is because the scematic of the Markforged CoreXY is not matching the Ender5 placement of the stepper-motors. So in the Markforged scematic the Y-stepper axis is facing upright while on the Ender5 it is tilted to the left, so the direction of movement is switched. There was also a commenter who switched the belt attachment points of the x-gantry from front to back and worked around with this solution.

endorphin3d, 2023-09-30:
I appreciate you sharing your findings here. I updated the docs for Klipper to change the endstop settings for correct homing:

```
[stepper_x]
...
position_endstop: 220
position_max: 220
...

[stepper_y]
...
position_endstop: 200
position_max: 200
...
```

Marlin seems to be similar but I haven't quite gathered the exact right settings from people, as I'm on Klipper myself.

powermandan, 2023-09-30:
I am waiting for a klipper Pad to turn up right now. Excited to increase the speed.

What sort of speed and acceleration figures are you using?

endorphin3d, 2023-10-13:
So my current bottleneck is no longer the physics of the printer but rather the rate at which my hotend can melt filament (I'm not using a volcano hotend) and the rate my part fan can cool the plastic. My acceleration is 3000 mm/s^2 and my infill speed is 120 mm/s but I'd bet can print faster than that if I upgraded my hotend and fans.

SirNukalot, 2023-10-19:
> So my current bottleneck is no longer the physics of the printer but rather the rate at which my hotend can melt filament (I'm not using a volcano hotend) and the rate my part fan can cool the [...]

Can confirm, i'm using a V6 hotend and it is now safe around 2500 mm/s^2 with wallspeed at 100 mm/s with very good printing quality and no layer shift. Beginning with 3500 mm/^2 the layer shift starts. Also tested only with 0.4 nozzle till now.

Setup:
- endorphin stage3 mod
- original stepper motors
- original mainboard v4.2.2 with TMC2209 stepper drivers
- runs on marlin 2.1.x bugfix
- dragonfly bmo hotend
- 4020 hotend cooler
- 2x5010 part cooling fans

mattswest, 2024-01-15:
I've seen this issue pop up twice in the last few days in the Ender5 subreddit.

I believe some setups will require `#define MARKFORGED_INVERSE` to be set in their Marlin config. It seems this was [added ](https://github.com/MarlinFirmware/Marlin/pull/26516)about a month ago.

This seems like it would be the solution for the issue [powermandan](https://github.com/powermandan) and [SirNukalot](https://github.com/SirNukalot) are seeing.

endorphin3d, 2024-01-15:
@mattswest you're a rockstar! Thanks so much for figuring this out. As I'm not running marlin anymore, is there anyone who's able to confirm if this fix works?

mattswest, 2024-01-16:
Unfortunately the [PR](https://github.com/Klipper3d/klipper/pull/6321) for similar functionality on Klipper seems to have stalled out.

I checked the Discourse but it looks like no one followed through with it.

endorphin3d, 2024-01-16:
Well I've never had an issue with Klipper and I think it's because of these two lines in my config:

```yml
position_endstop: 0
position_max: 220
```

I'll need to update the docs for that

closed 2025-01-11 by endorphin3d