endorphin3d/endorphin #2: Stage 3 - Missing EVA STL file

TristanPeers, 2023-10-13:
One of the EVA STL's is missing for [stage 3]( https://github.com/endorphin3d/endorphin/blob/main/docs/stages/stage-3.md )
The link to this STL is missing ->   [eva-endorphin_belt_grabber.stl](https://raw.githubusercontent.com/endorphin3d/endorphin/main/STLs/stage-3/eva-endorphin_belt_grabber.stl)
It is also not present in the repo, there are only 2 EVA stl's here -> [https://github.com/endorphin3d/endorphin/tree/main/STLs/stage-3](https://github.com/endorphin3d/endorphin/tree/main/STLs/stage-3)

Please can you add this file or advise where I can get it? Many thanks

endorphin3d, 2023-10-13:
Good catch, the file should be present now. Sorry about that.

closed 2023-10-13 by endorphin3d

TristanPeers, 2023-10-13:
Awesome - thank you so much!