endorphin3d/endorphin #4: Replace cantilevered bed?

kozross, 2023-10-24:
Is there any chance there could be a stage 4 to fix the bed being cantilevered? On a bed size this large, this is a _significant_ downside.

endorphin3d, 2023-10-24:
Believe it or not, that was exactly what I had in mind for stage 4. Here's a preview of what I had in mind:

[image]

Unlike similar projects I've seen on Thingiverse, it will only need one new rail and one new lead screw--either driven by a belt or by a second motor (there's tradeoffs for both and I'd love some opinions on which to pick). I've been busy with other work lately but if you or anyone else has the CAD skills to work on this, I can upload what I have so far.

kozross, 2023-10-24:
Basically, dual-motor is always better unless you lack the steppers on your board. Belt sync is _OK_, but it's always going to be second-best, especially given that this bed is fairly large. I think the stock board _may_ have dual-Z capability, but I'm not sure, so I'd say unless budget is a major concern, prefer two motors.

I definitely lack the CAD skills, so I'll wait patiently.

endorphin3d, 2023-10-25:
Yeah, it doesn't look like the stock boards have the second Z plug. The other upside of a belt is the two lead screws always stay in perfect sync and you don't have one side sag when the steppers are off (I've seen this happen personally).

kozross, 2023-10-25:
That's true, but to be honest, I'd probably replace the board with something like an M5P anyway, which _does_ have enough Z connections. The downside of a timing belt is that it has to be of a _very_ precise length (as you can't tension it), and thus, it's quite unlikely you'll find one in the right size without special ordering. A second motor and a board with dual Z steppers, however, is much easier to get hold of.

As someone who's had a timing-belt-linked Z printer, I can tell you that this kind of sag can happen there as well, and it's _much_ harder to fix. There's multiple printers without timing belts that solve this problem by literally crashing themselves into the ceiling (Prusa MK3, Sovol SV06, probably some others I'm forgetting), which I think that this printer would be capable of (assuming it has TMC2209s in it). 

At the very least, I would prefer having the option to dual-Z, with or without a belt. Single-drive Zs have given me issues with far smaller and less heavy beds in the past.

TheRealGennie, 2023-12-21:
My addition would be to delete or reposition the 2 shorter 2020 extrusions holding the z gantry (is that the correct term?), moving the gantry 20 mm further back, making room for a larger bed. With a narrower x gantry (due to rails instead of the wheels) that could make a great solution

endorphin3d, 2023-12-21:
@TheRealGennie that's a fantastic idea actually. Someone on Reddit recently [posted](https://www.reddit.com/r/ender5/comments/18chfdk/first_draft_on_ender_5_dual_z_project/) a dual Z project which didn't have the goal of enlarging the bed, but he did end up removing that 2020 extrusion extrusion and made his own adapter plate.

So to everyone in this thread, do we want to make dual Z _and_ larger print area the goal of stage 4? Just one tradeoff here: If we go down the path of having printed bed supports, that basically rules out the future possibility of adding an enclosure because those printed parts will sag.

kozross, 2023-12-21:
> @TheRealGennie that's a fantastic idea actually. Someone on Reddit recently [posted](https://www.reddit.com/r/ender5/comments/18chfdk/first_draft_on_ender_5_dual_z_project/) a dual Z project which didn't have the goal of enlarging the bed, but he did end up removing that 2020 extrusion extrusion and made his own adapter plate.
> 
> So to everyone in this thread, do we want to make dual Z _and_ larger print area the goal of stage 4? Just one tradeoff here: If we go down the path of having printed bed supports, that basically rules out the future possibility of adding a heat enclosure because those printed parts will sag.
> 
> (Vote with thumbs-up or thumbs-down)

I don't think the bed needs to be any larger honestly. One of the things that appeals to me about the Endorphin is that it's quite simple and makes minimal changes. A bed size mod feels entirely outside of that scope.

wildstar87, 2025-03-25:
I'm interested in having the mounts on the outside, at least on the front, to not lose build space, but options to move both to the outside would be nice as well.

Regarding the dual motor z, I happened to notice, after replacing the X-axis, with the extruder motor, that the old x-axis stepper is the same stepper as the existing z-axis stepper.  Since I went to the Sprite SE extruder, the old extruder mount (w/o extruder parts), is essentially the same mount that is used to mount the z-axis stepper as well.  

There is also this reddit post on adding a second z-axis driver to the Creality 4.2.2 board:
https://www.reddit.com/r/ender3/comments/qzxleg/i_added_a_5th_stepper_driver_to_the_stock/

So with minimal work, and just duplicating the BOM of the existing z-axis parts, a dual adjustable motor z-axis, seems to be fairly easy to do.

I know this would be extra cost, but I also found this plate on thingiverse, that could be sent out to online CNC services
https://www.thingiverse.com/thing:4677103.  That essentially does just that.