endorphin3d/endorphin #5: Ender 5 Plus-specific build plan/issues

tryptech, 2023-10-29:
Started working on implementing the Endorphin mod on my own Ender 5 Plus (E5P) and will document my findings and any potential issues here.

# Stage 1
This stage can proceed with no changes relative to the Ender 5 (E5) version.

# Stage 2
The usual linear rails used for E5P modification are 450mm millimeters in length so I will plan on using MGN12H at that length, specifically to follow al.

The left linear rail mounting will likely be fine. The E5P uses 2040 extrusion for more of its frame than the E5 so a different jig should be suggested.

The right linear rail will have a problem. The E5P uses two lead screws on the left and right of its bed in order to drive Z motion. The corners are guided by steel rods at each corner. These rods are fixed to 2020 extrusions that are connected to the side 2040 extrusions. If the right rail is mounted on the inside, the carriage would intersect with the 2020 extrusion below it, assuming the carriage has the standard width of 27mm. E5P linear rail mods only mount on the top or outside of the 2040 extrusions for this reason. Thus the mounting would have to be adjusted to other side of the extrusion.

# Stage 3
As is the case in stage 2, for standardization with other E5P mods, I intend on using another 450mm MGN12H rail.

If possible, I want to find a top mounted rail Afterburner/Stealthburner mount since I have a printed and working shroud + toolhead and would prefer to avoid buying parts to build a new printhead. That is to be determined before I proceed with this stage.

tryptech, 2023-10-30:
Stage 1 has been implemented relatively fine without a hitch, but I'm concerned about the relocated X motor. When moving along the Back-Left to Front-Right diagonal, the motor seems like it is at it's operating limit at ~160mm/s. This seems to be the case for both the 42-34 and the 42-40 motors.

The following videos are taken with the 42-40 motor in place.

https://github.com/endorphin3d/endorphin/assets/654389/c4334c4e-0964-44ba-bcb6-228cdd0e9e8d
https://github.com/endorphin3d/endorphin/assets/654389/edc848d4-d520-4fa6-bf88-27bb78abbed3

When moving along the Back-Right to Front-Left diagonal, the motor doesn't move in comparison.

https://github.com/endorphin3d/endorphin/assets/654389/2c2f52a2-671f-461f-adf0-ca2f36734391
https://github.com/endorphin3d/endorphin/assets/654389/1aeaa620-20c5-4bcf-8a20-1bf05b6a7c34

For now I've implemented a maximum speed of 150mm/s. This ends up being a downgrade to how I had my printer before relocating the motor. There I had an operating limit of ~185mm/s on Y-axis moves, with X-axis moves having a significantly higher limit.

endorphin3d, 2023-11-06:
Thanks so much for taking time to write this detailed review @tryptech. Are you seeing skipping on the X motor, or what are the signs you're seeing for the limit being ~160mm/s?

As for the rails causing collisions on stage 2, would 425mm rails solve the problem?

tryptech, 2023-11-06:
I don't believe the issue to be skipping, doesn't sound like it. I think the wheels aren't keeping up and are dragging and running on the extrusion. At least, that's what it sounds like. Other E5Ps can go faster than mine, so I'm chalking it up to these wheels being worn enough that they aren't suited for the task anymore. 
I'm going to proceed with stage 2 and 3 soon enough to leave the wheels behind, so it likely isn't a huge issue, but enough to note that a well tuned E5P may not necessarily see any speed benefits from stage 1. I expect Input Shaping and accelerations to have better results due to less mass being thrown around though. More time spent at top speed may be enough of a gain to justify Stage 1 for any E5P users who already have pushed the stock configuration as far as they can. 

The issue with the linear rail isn't the length of the rail, it's the width of the carriage. If you take a quick look at the [4020 extrusion on the top sides](https://grabcad.com/library/creality-ender-5-plus-1) you can see a 2020 extrusion attached to it on the lower inside part for the lead screws and steel rods. This poses an issue with mounting the rails on the inside because the nominal width of an MGN12H carriage is 27mm. No way it can fit at all. Hence the mounting needing relocation on the outside. 450mm should work fine as with all other E5P linear rail kits aside from that one core issue.

FilipeThePinho, 2024-05-14:
@tryptech do you still have the same issue with the X motor? And btw, which is the length of the belt you used?

Thanks!

geekyhawkes, 2024-08-12:
@tryptech what size belts did you use for the stage1 5+ update? Thinking of giving it a try but no idea what size belt to order.

LadyArtemis, 2024-10-09:
I'm in the process of making this upgrade now. I am doing all three steps at the same time. some things to note (and likely expected)

- upgrading the Ender 5+ is more expensive than the cost of the Ender 5/pro by approx $27 (liner rails run about $8 more than the ones in the guide. everything else should work)
- the 2040 rails required a different guide for placing the linear rails. I'm using [https://www.printables.com/model/224831-2040-mgn12-alignment-tool but I'm sure that any 2040 guild will work. ](https://www.printables.com/model/224831-2040-mgn12-alignment-tool but I'm sure that any 2040 guild will work. )
- I'm moving the motor to the back as it is with the standard guide so I will let you know what I see there. 
- I am using the Linear rail on outside and linear y switch on rail STLs that are in the issue linked above. 

Things I know will need to happen without even installing anything
- The klipper settings will need to change to 
`[stepper_x]
...
position_endstop: 360
position_max: 360

[stepper_y]
....
# Note: These may need to be 340 if using the EVA printhead
position_endstop: 350
position_max: 350'

I'm going to need to confirm those numbers and I am using the EVA so I will do that. 

I currently have parts on order and being shipped so I expect to be able to complete the upgrade in a few weeks.

geekyhawkes, 2024-10-09:
Great thank you! Certainly keen to hear your experience and lessons. I have
linear rails already so cost wise shouldn't be too bad but interested in
the tooth cable length required and any other observations.

Benefits post upgrade will also be of interest!

closed 2025-01-11 by endorphin3d