endorphin3d/endorphin #7: Placement of the rails, stage 2

TheRealGennie, 2023-12-20:
This mod looks very interesting, and Im planning to go this way. I do have a question (or probably a lot ;) 

Why not put both rails on the outside of the frame? 

My thinking is that it would increase the possible gantry travel by 50 mm. 
My plan is to add dual z (stage 4;) ) move the exixting z axis back, either to the main frame, which would give 20 mm, (or maybe even to the outside of the frame) and the standard hotend assembly have another 12 mm clearance to the front extrusion, This would allow for a larger bed.

endorphin3d, 2023-12-21:
@TheRealGennie Thanks for your interest in the project. I'll paste my answer to the same question from another user on Reddit (I need to add this answer to the docs, as obviously it's a common question)

> The reason the rails are like that is because of the asymmetrical tension the belt puts on the bearing blocks. A CoreXY has two belts, each pulling in opposite directions, so this isn't an issue. But on the Endorphin I wanted the belt tension to be pulling the bearing blocks toward the rail rather than away from it. If you look at a bearing block up close, I don't think it would take too much strength to rip it off a rail.

Maybe I should make a version of the right rail mount for users who want to put both rails on the outside. Personally, I would be curious to see how well a rail block stands up to the tension pulling it away from the rail. I may be fretting over nothing.

By the way I'm really hoping to find some time over the holidays to work on stage 4. Still open to suggestions on stage 4 if you'd like to weigh in here: https://github.com/endorphin3d/endorphin/issues/4

TheRealGennie, 2023-12-21:
Thanks for the answer. I really like this project!!
Im not an engineer, but I did study to be one for a year a LONG timer ago, so I remember a bit about forces. Anyway, (to me anyway) tension is the exact reason to put the rails on the outside. Belts cant push, only pull!! As they cross the middle of the printer, they can never push a bearing block away from the frame, only pull towards it, or towards the steppermotor. Hope it makes sense!! I tried to draw/speak through my thoughts in Explain Everything (https://expl.ai/3MFKKQV) if you are interested

endorphin3d, 2023-12-21:
Hey @TheRealGennie, thanks for the explainer video. The tension I was talking about actually wasn't due to the stepper moving the belt, but the tension that's on the belt 24/7 due to the belt feeding around the tensioner on the left side, but not having a similar belt feeding around a tensioner on the right side of the bed. See the yellow arrow:

[image]

But like I said, while the tension is there (you can pluck on the belts to feel it) it may not be _near_ enough to matter. I am a big fan of making the print area larger, so this may be worth doing.

tryptech, 2024-01-04:
Adding 2 cents, rail on the outside is exactly what is needed for Stage 2 on Ender 5 Plus as described in #5

endorphin3d, 2024-02-03:
And here's an [STL](https://github.com/endorphin3d/endorphin/blob/outer-rails/STLs/stage-2/linear-y-switch-on-rail.stl) for a Y limit switch holder that mounts in any of the rail's screw holes (with a T nut on the inside, same as all the other T nuts that mount the rail to the frame).

tryptech, 2024-02-03:
I'll give it a go this weekend and report back.

tryptech, 2024-02-05:
Two issues of note with the outside-mounted right-y-rail:

1. [image] Using the M5 bolt to mount extrusion to the part does not account for the space the head of the bolt takes up. The rail carriage is right up against the underside of the part, but the head of the bolt prevents it from sitting flush with the underside and making it impossible to screw in. I worked around this for now by installing an m5 heat set insert on the part and using a an M5x25 bolt from the top.

2. [image] The vertical alignment of the belt ends up hitting the rail with or without the bottom part. I will print the part at half height which should account for it.

sgayol, 2024-02-14:
" Using the M5 bolt to mount extrusion to the part does not account for the space the head of the bolt takes up. The rail carriage is right up against the underside of the part, but the head of the bolt prevents it from sitting flush with the underside and making it impossible to screw in. I worked around this for now by installing an m5 heat set insert on the part and using a an M5x25 bolt from the top."
And now I know why my AliExpress rail kit ended up lifting the X gantry about 6mm. 
[image]

endorphin3d, 2024-02-14:
Yep, I see the issue now. Thanks to both of you for testing. I'll try to design a fixed version this evening.

endorphin3d, 2024-02-15:
Okay, tell me if [this](https://github.com/endorphin3d/endorphin/blob/outer-rails/STLs/stage-2/right-linear-y-outer.stl) one works better. It now uses the inner M5 screw hole to mount to the X rail.

endorphin3d, 2024-02-15:
Okay @sgayol and @tryptech, let's see if [this](https://github.com/endorphin3d/endorphin/blob/outer-rails/STLs/stage-2/right-linear-y-outer.stl) one works better. It now uses the inner M5 screw hole to mount to the X rail.

sgayol, 2024-02-15:
> Okay @sgayol and @tryptech, let's see if [this](https://github.com/endorphin3d/endorphin/blob/outer-rails/STLs/stage-2/right-linear-y-outer.stl) one works better. It now uses the inner M5 screw hole to mount to the X rail.

Printed and installed. Fit is excellent. The only issue I have is that the tensioner I had for the x axis belt needed to be modified. Do you have one that's definitely compatible? It's incredibly helpful having one, especially with the longer belt path.

endorphin3d, 2024-02-17:
@sgayol What do you need to modify about the X tensioner?

sick4noise, 2025-01-01:
Thanks for all of your effort on this. I'm in the process of converting my E5 Plus.

I have another suggestion. With a bit of tweaking of right-linear-y-outer.stl, you could print this in on its side for better strength. It would also require less support. For example:
[image]

And with some chamfers on those edges you could potentially eliminate any need for supports.

DaKue77, 2025-01-01:
> Thanks for all of your effort on this. I'm in the process of converting my E5 Plus.
>
> I have another suggestion. With a bit of tweaking of right-linear-y-outer.stl, you could print this in on its [...]

Out of the same reason I did this for my Ender 5 Pro outer rail:
[image]

Shonky-Donkey, 2025-01-02:
I'll throw my hat into the ring here. This is what I came up with:
[image]

Haven't actually printed and tested this one yet though. Still waiting on parts to arrive on the slow boat from China for my build.
[linear-y-mount-right-ext.zip](https://github.com/user-attachments/files/18291804/linear-y-mount-right-ext.zip)

lastinfantry, 2025-01-18:
> Haven't actually printed and tested this one yet though. Still waiting on parts to arrive on the slow boat from China for my build. [linear-y-mount-right-ext.zip](https://github.com/user-attachments/files/18291804/linear-y-mount-right-ext.zip)

Did you have a chance to test your version already? I really like the approach. Probably going to print it next week.
Did you also model a  matching alternative part for the left side, with the fillets and all?

Shonky-Donkey, 2025-01-18:
I had already printed a version prior to this, which I decided was good enough, but the only differences with the one I uploaded was the gussets, and that I had taken the corner opposite the end stop striker off. 

I have no reason to believe it wouldn't work, and I would print the gusseted one if I did it again. The only thing I would add before I printed it, would be a zip tie hole, or other feature to route the x end stop harness through. 

I did beef up the opposite end a little, but not with gussets, though if I did it again I might try and come up with something more like what I did to the gussets side. You could probably go a bit wider than the rail block and gusset it there I guess. I am assuming the elbow is probably the weakest area of the part. 

Here's what mine looks like. It's also what I actually printed and used and it seems to work ok:

[image]

[linear-y-mount-left-outer_shonkyVersion.zip](https://github.com/user-attachments/files/18464902/linear-y-mount-left-outer_shonkyVersion.zip)

One other thing I might do, is slightly elongate the mounting holes for the 2020 extrusion so that there was some longitudinal adjustability: 

[image]

I did get my build going though. Working pretty well, but I think I used a too heavy grease for the linear rails, so it's a bit slower than it could be. I also made some bumpers for the ends of the MGN12 that also serve to make sure it's symmetrical when used on both ends: 

[image]

[EndStop.zip](https://github.com/user-attachments/files/18464881/EndStop.zip)

sick4noise, 2025-01-18:
+1 for elongating the mounting holes for the 2020 extrusion. Mine weren't even close and I had to use a drill to widen them.

My Stage 3 E5P build is almost complete. My latest issue is that the belts don't align with the standouts for the Vz printhead. On both the front and back the print part of the standout is too tall. I double-checked that I printed the correct bottom and back plates from the stage 3 folder. Anyone else encountered this? Maybe worthy of a separate issue?

[image]
[image]

sick4noise, 2025-01-18:
> +1 for elongating the mounting holes for the 2020 extrusion. Mine weren't even close and I had to use a drill to widen them.
>
> My Stage 3 E5P build is almost complete. My latest issue is that the [...]

Nm, I answered my own question :) It's because I'm using an MGN12 rail for the X axis.

endorphin3d, 2025-01-18:
Thanks for discovering this. I'll have to put in the docs that MGN12 won't work for the X axis.

endorphin3d, 2025-01-23:
Okay folks, I've listened to your recommendations and made right-side outer rail mount. Thanks to @sick4noise for the idea of side-printability and @Shonky-Donkey for the idea of tapering the under-rail portion rather than having a weak tab hanging out. Please test and let me know what you think!

[right-linear-y-outer.stl.zip](https://github.com/user-attachments/files/18515211/right-linear-y-outer.stl.zip)

sick4noise, 2025-01-26:
Thanks for posting the update! Not sure if there's an issue with the zip file but the part appears to be identical to the latest in the outer-rails branch.

endorphin3d, 2025-01-26:
Good catch @sick4noise. Sorry about that.

[right-linear-y-outer.stl.zip](https://github.com/user-attachments/files/18552007/right-linear-y-outer.stl.zip)

sick4noise, 2025-01-29:
Thanks @endorphin3d - that one looks much better!

I just completed my build and I'm in the process of tuning. My initial test cube looks decent, about the same quality as stock , which is to say it could be better. I was expecting better quality due to the switch from the stock extruder/hotend to the Orbiter/Rapido. But I digress. I'm trying to tune input shaping now and while the Y axis looks decent, X is a complete mess.

[image]
[image]

Has anyone else seen similar results? I've tried tightening and loosening the belts but it only seems to change the amplitude. I do hear lots of noise from the hotend when it gets above 120Hz, but I can't find anything loose other than the wire loom.

sick4noise, 2025-01-30:
Turns out I just need to increase the frequency range of the test. Then I found the dominant resonant frequency for the X axis:
[image]

I'm going to try the recommended shapers and run some tests but this looks promising.

sick4noise, 2025-02-03:
Spent all weekend trying to fix all of the input shaper issues and finally got to a good state. Apparently the first X belt I cut had a slight kink in it that was causing it to bind on the idler. It also took a few iterations to get the X belt lined up and even all the way around. I ended up stacking washers under the idlers to get the right height (could be replaced with a thinner version of the idler bottom).

[image]

After lots of tuning I was able to increase my acceleration to 6500 without ringing. Above that I start losing steps in the X motor, but I haven't tried increasing current as I don't have the specs on that motor. Though I've got five spares so at some point I'll will tinker with it. For now I'm quite happy with the performance. I printed a Benchy in 32 mins last night which would have taken ~60 mins with Klipper prior to this upgrade.

Having the rails on the outside seems to work well and I haven't lost any bed space. Since the X axis effectively fixes each of the linear rod mounts in the X direction I don't see how X belt could pull on the right linear rod carriage.

Thanks for all of your work on this! It's a wonderful contribution to the community.

endorphin3d, 2025-02-09:
@sick4noise thanks so much for testing this. So are you saying the groove in this new outside mount needs to be cut a little deeper so the X gantry sits a bit lower and you won't have to stack washers?

sick4noise, 2025-02-09:
@endorphin3d Yes, that could address the issue. Alternatively, you could print a thinner version of the center-idler-bottom.stl model and use that in place of the washers.

endorphin3d, 2025-02-10:
@sick4noise are your rails vertically centered on the extrusion? You can check using calipers or you can print a rail alignment tool like [this](https://www.printables.com/model/450861-mgn9-mgn12-linear-rail-2020-alignment-tools)

[image]

sick4noise, 2025-02-17:
Yes - I printed a rail alignment tool when I installed them.

wildstar87, 2025-03-07:
Is the outer rail mounting for the right rail finalized?  I'm looking at doing this mod, and going straight to Stage 2.  Is there a difference in the BOM between this, and the one on the main doc site, as it seems to only list the older inner rail mounting.

endorphin3d, 2025-03-08:
@wildstar87 we're just doing final tests now. You can get the files now on this [branch](https://github.com/endorphin3d/endorphin/tree/v1.1/STLs/stage-2). I just want to make extra sure the rail blocks still look fine after having the belt tension pull on them for at least a month.

wildstar87, 2025-03-30:
I printed both of the x-axis mounts on their sides, and used the original screws to secure them to the extrusion, and they ended up cracking right at the screw hole, both right in the center.  I think it was because of the depression in the screw hole, and screw head being larger than the depression, so when tightening it down, it flexed in the depression causing the crack, and it traveled right now the layer line, and cracked one of them clean through the middle.  I think it would be better to have this hole just be flat, instead of having that depression for a screw head, so the pressure is distributed evenly, or even a larger washer could be used to disperse the pressure over a larger surface area.  

I think printing them where the top area for the extrusion is facing the bed would solve part of this, but create another problem, especially with the right mount, as there is really nothing holding the 90 degree, other than that one layer, which is very thin.  

I think probably the best way to print for strength was the first example, where it is printed on it's end, with supports under the 90 degree portion.

ambarusa, 2025-04-12:
@endorphin3d just completed stage 2 with the outer rails. Everything turned out great using the default orientation in the slicer (one of the sides + supports). A quick observation: it would be ideal if the Y-endstop could be adapted to this version in a way that allows the back to support the rail instead of the 2020 profile.

[image]

jeffm89, 2026-09-18:
> And here's an [STL](https://github.com/endorphin3d/endorphin/blob/outer-rails/STLs/stage-2/linear-y-switch-on-rail.stl) for a Y limit switch holder that mounts in any of the rail's screw holes (with a T nut on the inside, same as all the other T nuts that mount the rail to the frame)
> 
> [image]

Hi @endorphin3d the link for the Y limit swich isnt working, is there another version anywhere? Thanks