endorphin3d/endorphin #8: clearance to acrylic enclosure

zCruuz, 2024-01-04:
Hi, 

since i saw the post regarding this on reddit i am curious to build it. Since back in the day i build a stupid Acrylic enclosure (https://www.thingiverse.com/thing:4314001) , i only have limited space.
Can you provide measurements or cad data for each step? I don't wanna get all neccessary parts just to realize, it won't fit.

[image]

endorphin3d, 2025-01-12:
Hey @zCruuz just cleaning up some issues. Did you ever get your question answered? The STEP file in the CAD folder should have everything you need.

zCruuz, 2025-01-13:
yes, took me some time to finally do the conversion. In the end i needed to dremel away parts of the enclosure to get proper access but nothing to bad.
Went for a HMG7 with Phaetus Rapido UHF and relocated the front aluminium extrusion.
Usable print bed size is now around 205mm since the Radpio sticks out quite a bit more than the stock mk8.

Next step will definetly be a bigger or relocated heatbed with active chamber heating.