endorphin3d/endorphin #11: Tension

Jonryux, 2024-03-31:
I am having a issue. Have been in days trying to figure out why the belts are not synced...

I am on stage 1 and used the extruder stepper as X axis.

The problem is like it makes a some kind of sound, maybe am missing steps or like everything is not really sync. I have been trying the tension on belts in different ways but still having trouble on the cube calibration. 
Here are reference pictures
[image]
[image]
[image]
[image]
[image]

endorphin3d, 2025-01-11:
That just looks like skipping to me. Try increasing the voltage on the X stepper a bit.

DaKue77, 2025-01-11:
Also make sure steps/mm are correct for the motors.

endorphin3d, 2025-01-29:
@Jonryux I was finally able to reproduce a similar issue, but it was only happening on a small circular part. This seemed counterintuitive to me because larger prints have longer travels to give more time to accelerate which would lead to a higher top speed. I messed with stepper current, `max_accel`, `max_velocity`, `minimum_cruise_ratio` and nothing seemed to do the trick.

After some googling I found people having issues with Arachne causing skipping on curves. I changed my slicer wall generator to "Classic" and the problem was solved. Let me know if this helps you too.

endorphin3d, 2025-02-14:
@Jonryux I just want to follow up on this. I was still able to create skipping when I increased travel speed. I played with the torque settings a bit to no avail, then I found [this page](https://www.klipper3d.org/TMC_Drivers.html) in the klipper docs. Basically, it seems stealthchop is the main culprit here.

> Tests comparing modes have shown an increased "positional lag" of around 75% of a full-step during constant velocity moves when using stealthChop mode

Then there's an additional potential cause:

> The TMC driver interpolate setting may reduce the audible noise of printer movement at the cost of introducing a small systemic positional error.

The solution is in the doc if you read on:

> For best positional accuracy consider using spreadCycle mode and disable interpolation (set interpolate: False in the TMC driver config). When configured this way, one may increase the microstep setting to reduce audible noise during stepper movement. Typically, a microstep setting of 64 or 128 will have similar audible noise as interpolation, and do so without introducing a systemic positional error.

So that's what I did and I'm able to achieve _significantly_ higher acceleration and travel speeds now. It's only _slightly_ louder when printing--mostly during long high speed travel moves--but it's not bad at all.

Here's my stepper config now (this is for an SKR mini v3). If you have any `stealthchop_threshold` lines in your config, delete it.

```cfg
[stepper_x]
...
microsteps: 128
...

[tmc2209 stepper_x]
...
run_current: 0.620
interpolate: False
...

[stepper_y]
...
microsteps: 128
...

[tmc2209 stepper_y]
...
run_current: 0.650
interpolate: False
...
```

fraserw, 2026-01-04:
@endorphin3d  fantastic settings. Thanks for that. The noise without stealthchop was unbearable. But your settings have brought that noise mostly back down and the accels are much higher.