endorphin3d/endorphin #13: Endorphin on Ender 5 plus

Stenli37, 2024-10-20:
Hi. I did stage 1 and 2 simultaneously and will do stage 3 eventually. So I got MGN12H 450mm linear rails and 5 meter belt, everything else stayed the same as in the guide. Firstly, I printed all the parts without any changes, but after disassembling everything I discovered that the rail doesn't fit on the right inside because of the Z-axis rods being mounted to an additional side rail/gantry.  So I removed it and the printer worked ok enough to print a modified version of the y mount (https://www.thingiverse.com/thing:6803371/files) (idk how to upload stl files here, sorry). I have it printed and everything fits nicely. Other than that, I didn't move the endstops, since there was no need to.  I will upload some vids later.
As for the firmware, I used this - https://github.com/InsanityAutomation/Marlin/tree/CrealityDwin2.0_Bleeding with this config, because I can use my stock screen, without any changes https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-5%20Plus/CrealityV1%20(RELOADED%20UI). In the marlin config file I had to enable the inverse markforge otherwise the printhead would just run in the opposite direction.

#define MARKFORGED_XY
//#define MARKFORGED_YX
#if ANY(MARKFORGED_XY, MARKFORGED_YX)
 #define MARKFORGED_INVERSE  // Enable for an inverted Markforged kinematics belt path
#endif

sick4noise, 2024-12-27:
@Stenli37  Can you post a pic of your modified Y-mount on Thingiverse? I'm interested in using it for my E5P but not quite sure how it mounts.

Shonky-Donkey, 2024-12-31:
> @Stenli37 Can you post a pic of your modified Y-mount on Thingiverse? I'm interested in using it for my E5P but not quite sure how it mounts.

Something like this, I suspect: 
[image]

sick4noise, 2024-12-31:
> Something like this, I suspect: [...]

It seems so obvious now :) Thank you!

Stenli37, 2025-01-07:
Hey, finally finished the stage 3. It took me some time, since I converted to mircoswiss direct drive and had some trouble finding a linear rail mod for it, together with a new printhead. Attached is a video of the final thing and a picture of the y-mount, as requested above. 
So for the linear rail I used this - https://www.thingiverse.com/thing:5002103 - no additional modifications, didn't move the y-endstop.
For the printhead - https://www.thingiverse.com/thing:5178108 - however, there were a couple of things. The connector of the BLtouch on the e5+ is bigger/bulkier, so I had to drill and melt a piece from the "StealthSwiss_mounting_plate_universal" so that it would fit. The top left hole on the direct drive plate, to which the linear rail adapter is mounted, is also covered by this plate, so I drilled a new one. I also had to move the X-endstop to the right a little bit, since the nozzle would start printing the cleaning line a few millimeters off the plate. I did this by trial and error, so I cannot say exactly how much. I also used the y-endstop file from stage 2 to lift the x endstop a little.

Some time ago I bought an x tensioner from aliexpress, which, as I later found out, is normally mounted sideways and is for Ender 3, I think. But anyway, I managed to make it work - had to melt a bit of the corners from the endorphin mount and also use 3 washers to fill the empty space between the rollers on the mount. I will probably buy/make a better one in the future.

I know the belts on the y-axis should go under the mount, but I found that out much later and am now too lazy to disassemble everything.

The other mods I have: 
https://www.thingiverse.com/thing:4825660 - bracers 
https://www.printables.com/model/116582-ender-5-plus-crossbar-variable-relocator - crossbar relocator
https://www.thingiverse.com/thing:4871661 - toolholder
https://www.thingiverse.com/thing:4255482 - cable guide tower
https://www.thingiverse.com/thing:4165428 - bed cable strain relief

I have also prepared some of the bed spacers from here, just in case if my bed wasn't high enough for the nozzle-
https://www.printables.com/model/563494-ender-5-micro-swiss-dd-linear-rail-adaptor/files

Also, if you convert to direct drive don't forget this, very useful -https://www.thingiverse.com/thing:3593964

[image]

https://github.com/user-attachments/assets/ca2d533a-16fd-49e7-885e-afe3b562dd1a

endorphin3d, 2025-01-11:
Thanks @Stenli37 for all the info! I think that `MARKFORGED_INVERSE` line was what's been missing for people using Marlin so I added that to the docs. I'm going to close this issue and let the convo continue on issue #14 about putting the benefits/drawbacks of putting rail on the outside on the E5+.

closed 2025-01-11 by endorphin3d