endorphin3d/endorphin #14: Endorphin on a 5+ HIGHLY recomended

GarthyJBI, 2024-11-22:
Well i cant reccomend this upgrade enough for the 5+

I have wanted to upgrade my machine with linear rails since I got it in March but just havent got around to it but the more hours it did the more those stupid wheels were wearind away and just leaving mess everyxtime I did a print and the amount of adjustzments I had to do everytime before I set of a print was just getting on my nerves so I looked around for an upgrade and came across this recomended in a Reddit post.

So i looked at the cost and figured it was worth a try cause apart from the linear rails which I was going to need no matter what I did I had practically everything else laying around from tinkering with 3d printing over the years just bought a 10m belt and the rails and away I went.

There are some differences between doing this on a 5 and a 5+ the main one being when fitting the right side linear Y axis rail the stabilisers for the bed are bolted to the frame and so you cannot fit the rail so I decided to just unbolt them and they are sturdy anywaysd so didnt bother finding a way to reattach them.

I also decided that for now to keep costs right down I didnt want to do a hotend upgrade so printed a new fan peice for the original hote end and found a linear rail holder for it on thingiverse and so far so good.  

So after running through all calibrations and speed testing Im am currently reprinting a moon lampshade that I printed last week and it took over 58 hours then but now Its at 95% and has only taken 16 hours and the quality is better than the 58hour one.

Thank you very much for the MOD well worth it for such an uprade which takes every little effort and cost.

[image]

aaron-harvey, 2024-12-06:
I used [this tensioner](https://www.printables.com/model/885298-endorphin3d-ender-5-y-axis-belt-tensioner) and have been pleased with it

Shonky-Donkey, 2024-12-31:
What about this to solve the Z rail issue?
[image]

[image]

[linear-y-mount-right-ext.zip](https://github.com/user-attachments/files/18279010/linear-y-mount-right-ext.zip)

I haven't tested it yet, but it's what I plan to try. I've attached a step file if you want to be the guinea pig.

endorphin3d, 2025-01-11:
Hey @GarthyJBI thanks so much for the tip on unbolting the vertical rail holders. I can add that to the docs if a few other people confirm that it's still 100% stable or I'll accept PR's for anyone who wants to design a replacement that bolts on the underside of the Y rail.
As for the tensioner, I finally got around to adding one. See my comment [here](https://github.com/endorphin3d/endorphin/issues/9#issuecomment-2585400969).

endorphin3d, 2025-01-11:
@Shonky-Donkey great minds think alike. I have a _very_ similar design on a branch [here](https://github.com/endorphin3d/endorphin/blob/outer-rails/STLs/stage-2/right-linear-y-outer.stl). If you'd like to be the tester, let me know.

Stenli37, 2025-01-17:
@johnscrub For the X belt I bought a total of 5 meters length, and I've used 2,4m. So if you buy 3m, you will be good. I kept the Y belts stock, and I suppose they should be good if you put them under the mounts as intended. They aren't long enough if they go above the mount like mine, so I made these "holders" for them, that you can see in the picture/video.
And yes - 3 MGN12H 450mm rails.

lastinfantry, 2025-01-18:
I think you can get away with shorter rails, at least for the X-axis.

I got a really good deal on 350mm rails and I am currently printing the parts for the Endorphin mod (for my Ender 5 Plus). I'll update here sometime next week once I tried installing it with my short rails.

I already have a rail on my X-axis. On the left side I mounted a block on the 2020 extrusion to avoid the train sliding off the rail in case of skipped steps (instead of using the little plug on the screw hole of the rail). On the right side I kept the stock endstop switch, still mounted on the extrusion.
This way I only lost very little print volume despite the short 350mm rail. With 400mm rails you should be able to use the full bed.

But if you have to pay full price, just get the 450mm ones to be sure. This way you can probably mount physical endstops on the rails themselves (mine came with "plugs" for the screw holes) instead of on the extrusions, much cleaner and a little less weight.

endorphin3d, 2025-01-23:
Okay folks, I've listened to the recommendations of a few people in issue #7 and made right-side outer rail mount. Please test and let me know how it works.

[right-linear-y-outer.stl.zip](https://github.com/user-attachments/files/18515231/right-linear-y-outer.stl.zip)

johnscrub, 2025-01-23:
@Stenli37 would you be able to share your config changes n for klipper? I've got stage one done but the docs are only for E5 or e5pro. Thanks

Stenli37, 2025-01-24:
@johnscrub I'm using marlin with stock board and screen. You can see the version and everything in the links in my post.

johnscrub, 2025-01-24:
> [@johnscrub](https://github.com/johnscrub) I'm using marlin with stock board and screen. You can see the version and everything in the links in my post.

Thank you, my GitHub etiquette is not good. I saw way after I posted. I do apologize

johnscrub, 2025-01-26:
Can anyone that is using this on an e5plus and running klipper... Share what changes they made to there cfg? Or there complete printer cfg? Tia

CantElevate, 2025-01-29:
> Can anyone that is using this on an e5plus and running klipper... Share what changes they made to there cfg? Or there complete printer cfg? Tia

https://endorphin3d.com/docs/firmware#klipper 

These are all the changes required for endorphin3d in klipper. everything else should be the same as it was before you changed the physical parts of the printer.

I use Klipper, however my setup is about as individual as they come.   

The only real difference with the plus in regard to endorphin is the dimensions.

ingramcl, 2025-01-30:
@GarthyJBI would you be able to post the fan and linear rail files you used for the stock hot end.  I am about to make the jump with my 5+ and like you won't be upgrading the hot end just yet.  I am going to start printing the parts tomorrow and the ordered parts should be here soon.  Thanks!

khisanthax, 2025-02-08:
The only changes in klipper were the kinematics line?  I had to swap the pin direction on the y axis but I still get no trigger on x.  Any other changes?

khisanthax, 2025-02-08:
> The only changes in klipper were the kinematics line? I had to swap the pin direction on the y axis but I still get no trigger on x. Any other changes?

I over tightened the screws on the idlers on the x gantry binding movement completely.  I had washers on the tops and bottoms, but assumed that I wouldn't have to worry about over tightening.  I either used the wrong washer or was just wrong about over tightening.

Kikkero, 2025-03-16:
> [@johnscrub](https://github.com/johnscrub) For the X belt I bought a total of 5 meters length, and I've used 2,4m. So if you buy 3m, you will be good. I kept the Y belts stock, and I suppose they [...]

The instructions for stage 3 say to use a MGN9H rail for for the X axis.  Is there a reason why when doing this mod on the E5+ I should go with the MGN12H for the X axis, instead?  I don't know too much about the differences between these rails, but the MGN12H rails look to be a bit beefier than the MGN9H ones and the carriage looks different.  I suppose the beefed up rail would be good for rigidity, but wouldn't this cause problems with compatibility, weight, and/ or clearances?  

I have everything here to do stage 1, which I plan on cracking into here in the next few hours, and I have the stage 2 rails on the way.  I'm still waiting to jump on stage 3 until I make a decision on what I want to do with the tool head.  I've already got a Spider 3 hotend and Sprite DD extruder (not the sprite pro kit) installed which are plenty to do what I need (I bought this thing in a state and I just needed a drop-in fix quick with minimal surprises.  The only thing I really (really, really) need as far as the basic toolhead performance goes is better part cooling to keep up with all the extra heat I'm pumping out.  Or am I just completely off base and this configuration will be totally inadequate and become a major bottleneck for me once I'm throwing that head around at the upgraded speeds?  And how will this affect which rail I need (if at all)?

yourzero, 2025-04-03:
> [@GarthyJBI](https://github.com/GarthyJBI) would you be able to post the fan and linear rail files you used for the stock hot end. I am about to make the jump with my 5+ and like you won't be [...]

@GarthyJBI or @ingramcl - Do either of you have these files? I'm in the same boat - going to Stage 3 but with the stock hot end.

Thank you!

CantElevate, 2025-06-13:
When I originally did the stage 3, I went looking for a mount that made things work. this one  seemed to replace the stock mount with one that works with the rail and front connecting belts of the stock motion system 

X_Axis_Linear_Rail_Mount_final.stl  
https://www.thingiverse.com/thing:3782865/files

CantElevate, 2025-08-26:
> I am looking into the endorphin3d mod for my ender 5 plus. However, I'm not seeing a lot of info on changes to the bill of materials (particularly the rails) for the ender 5 plus. I'm assuming that the BOM specs are for the standard ender 5.
> 
> Does anyone have info on the exact changes needed to the BOM for the ender 5 plus?

I used 3 x **450mm** mgn12h rails for my ender5 plus conversion, but knew early that the mercury one.1 was a likely project in the future.  I don't recall any other changes to the Endorphin bom for the conversion.