endorphin3d/endorphin #15: Stage 3 Linear rail type

ambarusa, 2024-12-15:
Hello,

Before diving into this stage, I would be interested to know what is the reason of using MGN9H type of rail for this rail?
My case is a bit specific, I have an easy access to MGN12C rail, that is the recommended rail type for EVA printhead. 

Apart from saving extra weight with MGN9 rail, is there any reason why we should use it?

endorphin3d, 2024-12-15:
MGN9 was chosen simply because it weighs less than the 12. MGN12 should work fine, it's just going to weigh more so your acceleration and jerk settings will max out at lower values.

ambarusa, 2024-12-15:
Makes total sense,I’m fully convinced. It was good though to clarify that it’s not absolutely obligatory.

closed 2024-12-15 by ambarusa

ambarusa, 2025-03-10:
@endorphin3d FYI, with this approach I bought MGN9C rail type for even less weight, and used the corresponding STLs from VZPrinthead and your custom ones.

_"Make sure you print the MGN9H version of the carriage top (assuming you bought your rails from the links above) rather than the MGN9C version."_ - This note made me afraid that the C type will not be compatible, but so far no issues at all.