endorphin3d/endorphin #16: tension screw video wrong

lwkalis, 2025-02-11:
The video for installing the tension screw show the m5 going in to the carrier with the nut close to the pulley. when tightening the bolt it runs into the belt. Bolt should be inside and nut should be on the knob.

endorphin3d, 2025-02-11:
Good catch. Thanks for noticing. I'll close this issue when it's updated.

DaKue77, 2025-02-11:
> The installation video for the tension screw shows the M5 bolt being inserted into the carrier with the nut positioned near the pulley. However, when tightening the bolt, it interferes with the belt. To prevent this, the bolt should be placed inside, with the nut secured on the knob.

During my mod I faced the described issue and so I remodeled the knob to be thicker.
@lwkalis , your approach is so much easier and also logically more straightforward! Very good point! 👍🏻

endorphin3d, 2025-05-27:
Fixed in 1.1