endorphin3d/endorphin #18: Ender 5 Pro uses M4 bolts and tnuts

wildstar87, 2025-03-21:
I'm a bit confused on one part of this.  In Stage 1, the instructions say to reuse the M3 bolts and tnuts, when assembling the Front Corner Assembly.  The screws I have in my Ender 5 Pro, are M4 bolts and tnuts.  They use an M3 allen wrench, and the thread diameter measures out at a little under 4mm.  Is this possibly a typo?  I'm wondering if the holes in the models are going to be big enough for me to use my existing M4 bolts/nuts or if I will have to buy M3 bolts and nuts to replace these, or just drill out the printed parts, and whether the screw holes have enough material for strength for me to do that?

Looking at a parts list for the Ender 5 Pro, it seems it definitely does use M4 bolts/tnuts.

https://www.3dmaterial.cz/files/shop/4-2022/creality/user-manual-ender-5-pro-en-v.2.1.pdf

wildstar87, 2025-03-25:
Probably want to update the docs, if/when the instructions are updated for v1.1.  I did check the models, and the holes definitely seem big enough for M4, so it definitely looks like a typo

DaKue77, 2025-03-25:
@wildstar87 I can also confirm M4 to be the original bolt size, at least on my Ender 5 Pro model.

endorphin3d, 2025-05-27:
Thanks for this. It's been fixed now.

closed 2025-05-27 by endorphin3d