endorphin3d/endorphin #19: Driving X-mount difference between v1.1 and main?

wildstar87, 2025-03-24:
I started printing this mount out, and noticed that the v1.1 version has a notch in the top surface, vs a smooth top on the one in main.  What is the reasoning for the notch?  I would think this would add a potential weakness to the part.

endorphin3d, 2025-03-25:
That notch is for you to access the grub screw on the gear with a hex key for better alignment of the belt. Considering how much this mount was beefed up in other places for 1.1, I don't think this notch will be an issue.

wildstar87, 2025-03-25:
I see, I'm trying to swap over the extruder motor to X-axis, but the shaft is shorter, and round.  I was able to remove the gear, and bevel out a flat spot on the shaft, to use the existing timing gear, hopefully it won't need to be moved up much further, otherwise I may have to get another stepper with a longer shaft.