endorphin3d/endorphin #21: Suggestion for Marlin FW / layer shifting

tocsa120ls, 2025-03-29:
The mod works perfect (Ender 5 Pro + v4.2.7 mainboard + Stage1 so far with me) but the firmware configuration can give headaches. Just want to drop a few tips, and the issues I've experienced:

* forget the mainline Marlin firmware. It is about 15 months behind the _bugfix-2.1.x_ branch and so many things were fixed, especially regarding CoreXY kinematics. Do not use the mainline.

* my alterations to the (bugfix) firmware:

* uncomment these 2 lines in _Configuration.h_
```
#define MARKFORGED_XY
#define MARKFORGED_INVERSE  // Enable for an inverted Markforged kinematics belt path
```

* if you have swapped the X stepper with the extruder stepper, you can and probably need to up the current of the motor since we removed the weight from the X gantry, but we added 4 pulleys to the belt path - for me this was causing the random layer shifting issues in one direction

* adjust these 2 values in _Configuration_adv.h_
```
    #define X_CURRENT       650        // (mA) RMS current. Multiply by 1.414 for peak current.
```
crank it up from 580 - I matched the Y_CURRENT which is 650

* similarly, the extruder stepper is small now 
```
    #define E0_CURRENT      580
```
reduce the current to 580

That's it.

jjn-9306, 2025-05-02:
I will try this after the current print finished. I swapped the X and E motors but did not change anything else. I am noticing when the print head travels along the Y = -X line, the X motor makes a loud buzzing noise. I'm guessing this is because the combined X/Y motion in that direction is forcing the motor to move faster than it can safely keep up?

I'm hoping increasing the V_ref and changing the current will fix the issue.

endorphin3d, 2025-05-02:
Yeah so I'm pretty certain that sound comes from the slight positional inaccuracies that come from stealthchop. My theory is one stepper corrects the other and causes some jerking motion (leading to vibration = sound).

The reason for this theory is when I turn off stealthchop and increasing my microsteps to 128. Basically getting increased torque and positional accuracy while approaching the same level of quiet as stealthchop

Unfortunately the 4.2.7 board has stealthchop hard wired so you'll need to do some soldering or change boards.