endorphin3d/endorphin #22: Requesting a version that uses 5mm bore pulleys

wildstar87, 2025-04-11:
The 3mm bore pulleys that seem to be available, I have found, don't have enough effective clearance between the bearing core on the outer, so even with washers on both sides, tightening these down, will cause binding, increasing friction dramatically.  There seems to be much fewer options, since this is a more uncommon bore size.  I know that the 3mm bore was decided on to reuse the stock carriage holes without modification in a Stage 1.

I went directly to Stage 2, and trying to keep the pulleys loose enough that they don't bind, while keeping them from wobbling is difficult, and can loosen up tolerances just by being used, I think because the wobble wears on the 3d printed parts, causing them to loosen up, even when using nylock nuts.

The more common 5mm pulleys have much more selection, and would probably be easier to ones with bearings that had proper clearance, that wouldn't bind when tightened down.

Would be nice to have a version that used 5mm bore pulleys in addition to the existing 3mm design.

Zoolander06, 2025-04-21:
Hello,

You could very easily design some spacers to put inside your 5mm bore pulleys.
I had old belt tensioner reusing the original pulleys with that kind of spacer to use M5 bolts instead of the genuine M8, and it worked perfectly.
The issue I have with 3mm bore pulleys is that washers are to big and prevent me to tighten screws correctly, perhaps thats what causing your issues. For that, perhaps some printed washers 5x0.6mm would help.

endorphin3d, 2025-04-22:
Yes I like @Zoolander06's idea of making an adapter that would slot into a 5mm bore. That's far more preferable to redesigning almost every part in the system plus modifying all the docs. If anyone wants to design that, you can make a PR and I'll review it. Please include a STEP file so I can integrate it into the project.

Elucubra, 2025-05-15:
Hola. 
Is there a good reason why the holes can't be redrilled to 5mm? Trivial with a bench drill adapter or similar.

nocturneop15, 2026-10-16:
If you are still interested, I made variant of strenghtened endrophin, where I use 5mm aluminim standoffs. It greatly enforces overall rigidity of the system and allows to use more standard 5mm bore pulleys right of the shelf:

Right assembly: 

[image]