endorphin3d/endorphin #24: Unparallel belt travel

Zoolander06, 2025-04-21:
Hello, I noticed that the hybrid belt travel slightly not parallel to the Y axis. I think it would be better that everything is parallel, for better accuracy.
It would be solved by moving the stepper motor, and the idler 2mm to the left (with a 1.5mm thick belt).

Also, the travel along the X axis is only parallel with the genuine idler, with the shitty tensioning system. If you use the Endorphin tensioner, with the 20 teets pulley, the belt is not parallel anymore. I made some modifications on the tensioner mover so it can reuse the original 28 teeth idler, but I hadn't tested it yet...

Am I wrong ?

Joris

endorphin3d, 2025-05-04:
Would you mind posting a few photos? I can't quite visualize the issue you're seeing.

Zoolander06, 2025-05-16:
Hello,
Well, you can easily see the issue on the cad model : 
[image]
Let's forget the belt, which is only there for looking good since it doesn't even touch any of the pulleys, and you can see that there's about 3.5mm between the idlers on the X gantry, and the driving pulley. Same with the idler of the corner assembly. So, the X belt is not running perfectly parallel to the Y axis, which, I believe, can lead to unequal tension depending of the Y position.
This part is easy to correct, you just need to move both the motor and the front corner idler so the belt run perfectly parallel (I would do it myself but since I don't have the design history in the step file, it's a real nightmare).

There's a similar problem along the X axis, which is already there on the original Ender 5. I think I could solve it by putting some spacers on the X carriage.

I don't know exactly how much it impacts the precision of the printer, but I think that, since it's a simple fix, it's always better...

I hope this is clear, english is not my main language, so it sometimes difficult for me to explain...

Joris

Elucubra, 2025-05-21:
Joris, 
Calling the work of someone who has spent a lot of time and effort, and sharing it for free, shitty is disrespectful at best, I would call it disgraceful.

Please indicate where is this lack of parallel

[image]