endorphin3d/endorphin #28: Debris from the belts around the tensioners

ambarusa, 2025-06-11:
Hello,

After a while of using the X tensioner, I noticed some debris from the belts due to the touching the sharp edges of the rails. I can definitely imagine that this affects the print quality as the belts are not completely moving free. 

I can also notice that even the belts on the Y axis are touching the edges of the rails.

Does this occur to you?

[image]

ambarusa, 2025-06-11:
I would also be interested if you think necessary to add washers to the X tensioner. Even though the pulley is in horizontal position, I can see that it never lifts up or down to touch the plastic. For the Y ones, the guide clearly mentions that washers are not so important. It is really hard to push them on both sides of the pulley anyway.

fraserw, 2026-02-08:
I also have these issues. Though my solution was very careful alignment of pulleys and belt ends such that the belts never touch the metal, at least as far as I can see. Though I am getting bad VFAs, so maybe there is rub somewhere.